mike1pol/octoprint_macro #7: Multi Line Gcode Can Be Extremely inconsistent!

Dracrius, 2020-07-22:
If a line of gcode ends with a `,` or `;` it will inconsistently execute. I had this set
```
G91;
G1 F3000 Z20 ;
G28 X Y;
G90;
G1 X0 Y235 F1000;
M84
```
And it was all over the place on which lines it would us, sometimes only running the first 3, other times running only the third etc.

It appears to work as expected if you enter 1 line of gcode per line with no punctuation or comments.

mike1pol, 2020-07-26:
1 line per 1 command.
for comments use ``;;`` on new line or after gcode line

mike1pol, 2020-07-30:
do it work?

dannieboiz, 2020-08-06:
I'm also having this issue, trying to move X, y to 117 and Z to 0

I've tried G0 X117, y117, z0 and G0 X117; y117; z0 none works can you please give an example of what it would look like?

meltonpieman, 2020-08-06:
dannieboiz
I have used multiple commands on one line with no seperators, such as
  G26 B60 H210 L0.3 R1
and had no problems
David

dannieboiz, 2020-08-06:
Thanks, issue was me... In terminal it isn't case sensitive but in this plugin it is, G0 X117 G0 Y117 G0 Z0 works.

G0 X117 isn't the same as G0 x117 it seems LOL

mike1pol, 2020-08-12:
need to convert all commands to uppercase before executing a command

commit 3e35f79f (2020-10-20): close #7, close #10
closed 2020-10-20 by mike1pol