mike1pol/octoprint_macro #17: Commands sent uppercase in terminal

slvrsky07, 2021-01-17:
Using octoprint, with this command in the command box: `M118 //action:camon` It will send the command as all uppercase. This does not work with the action command plugins since they need to be sent as shown. Is there anyway to change that in the macro to send in both upper and lower case?

This is the terminal output shown: ```
Send:M118 //ACTION:CAMON
Recv: //ACTION:CAMON
```

David-Maisonave, 2021-04-27:
Posting more details on this issue:
### **Describe the bug**
**Summary**
The plugin does not adhere to Terminal Auto Uppercase Blacklist.
**Details**
Currently all GCode is getting converted to uppercase.  This causes action commands on M118 to fail.
Example:
M118 // action:disconnect
The above case sensitive command fails to disconnect the printer when issued from the macro.
OctoPrint has an Uppercase Blacklist option, which is prepopulated with "M117, M118".  That allows those two commands to succeed.
The Macro plugin should avoid making uppercase any line that has a command in the uppercase blacklist.
Or better yet, just send all commands as-is.
### **To Reproduce**
Steps to reproduce the behavior:
1. Add `M118 // action:disconnect ` to a macro.
2. Run the macro
3. Printer does not get disconnected, and the terminal window displays GCode in uppercase.
### **Expected behavior**
GCode in the uppercase blacklist should be sent as-is.
To simplify things, all commands should be sent as-is.
### **Versions**
 - OctoPrint: 1.5.3 and 1.6.0
 - Browser [Firefox]
 - Version [0.3.0]
### **Additional context**
The Uppercase Blacklist can be found on OctoPrint under Settings->Features->Terminal Auto Uppercase Blacklist.

David-Maisonave, 2021-04-27:
_FYI:_
I downloaded the source and remove the toUpperCase commands in file **octoprint_macro\static\js\macro.js**.  They were located in line 36 and line 40.
I tested it, and the action commands work with this small change.  

However, if the user enters other commands in lowercase, those commands do fail.  I was hoping that OctoPrint would make the uppercase conversion, since it's already doing that for the terminal input.

I hate reinventing the wheel, so I'll look to see if there's a way to send the command where the terminal input logic gets used.

David-Maisonave, 2021-04-27:
I found that by using a Terminal Tab hook, the macros will get the uppercase logic handled automatically, without have to do a toUpperCase call within the Macro Plugin.
I added the fix in pull request #22.

Please ignore #21, which I was doing on a fork, instead of a branch.

closed 2022-01-09 by mike1pol