mike1pol/octoprint_macro #26: Version 0.3.1 does not update version number, sticks in infinite update loop

AlAmantea, 2022-01-09:
**Describe the bug**
A clear and concise description of what the bug is.

Version 0.3.1 does not update version number, sticks in infinite update loop

**To Reproduce**
Steps to reproduce the behavior:
1. Update to 0.3.1
2. upon restart, version still reports as 0.3.0 and asks to update again
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Version should automatically update to 0,3,1

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - OctoPrint: [e.g. 2.4.1]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 0.3.0]

**Additional context**
Add any other context about the problem here.

ejhoness, 2022-01-10:
mee too

mike1pol, 2022-01-14:
fixed

closed 2022-01-14 by mike1pol