mike1pol/octoprint_macro #28: not processing in order entered

htrabbit, 2022-01-30:
**Describe the bug**
when i first boot up, the home all macro fails

[image]

when i press it a second time i get the following 

[image]

**To Reproduce**
Steps to reproduce the behavior:
create this macro seen below

[image]

**Expected behavior**
i expect a G91, a home x0 ,y0
then a home Z0, then a G90

**Versions (please complete the following information):**
 - OctoPrint: 1.7.3
 - Browser [Edge 97.0.1072.76]

aliasmeum, 2022-02-28:
Hello,
I am experiencing the same behavior, gcode lines in the macro are executed in random order. I have already crashed the nozzle in the printbed testing out this plugin. I love the plugin, but this behavior is dangerous and not acceptable.
This plugin needs to be removed from Octoprints plugin repository or at least a clear warning added until it is fixed.
I really hope you can get this sorted out, because this would be one of the most useful Octoprint plugins if it worked. Thank you for your great work on the plugin.

aliasmeum, 2022-02-28:
Hello again,
A workaround is to use the OctoPrint Gcode Macros Plugin by Charlie Powell to set up the macros you want. Then use these in the Sidebar Macros plugin.

In Gcode Macros plugin:
[image]

In Sidebar Macros plugin:
[image]

Users should still be warned about Sidebar Macros behavior!

levpopov, 2022-04-06:
+1, the default behavior of sending commands in random order is extremely unsafe, and worse yet not made known to the user. This plugin should be blacklisted from Octoprint plugin manager until this is resolved.

oishii-code, 2023-01-29:
I've noticed the same issue. Thanks for the workaround @aliasmeum 🙌